CrealityOfficial/Creality_Sonic_Pad #3: Creality Ender-3 S1 CV-Laser Module - Ender 3 S1

LA3QMA, 2022-10-18:
I'm working with the Sonic Pad so that i can use the Creality Ender-3 S1 CV-Laser Module.

So far i have managed to configure printer,cfg so that i can make macros to support the laser.

I got two problems:

Issue 1. Disconnecting the flat cable from the hotend the Sonic pad goes into safe mode as it should. But to use the engraver i have to disable this warning and disconnect the cable. It says the temperature is to low as its trying to measure from the sensor that now are removed.

It could be solved changing the "min temp" i the extruder config but then i need to know what value it's measuring if the sensor is not connected.

When issue 1 is solved i need access to the Expansion interface 24V PWM for the Ender 3 S1

Issue 2 for my macros to work with the CV-Laser module i need to know the pin definition for Expansion interface 24V PWM.
The part cooling fan is PA0. What is it for the Expansion interface 24V PWM ?
Or could i use the PA0 to control the CV-Lasermodule if i make a proper cable?

When all this is ok i can have a look to make a UI for the Sonic..

Creality-Ralf, 2023-01-04:
The port definition is PC0

LA3QMA, 2023-01-04:
Thank you. I'll try to make a usable macro/config for the laser engraver.

chuckxxxl, 2023-02-22:
> Thank you. I'll try to make a usable macro/config for the laser engraver.

Any Updates?

Creality-Ralf, 2023-03-13:
We will support the laser engraving function of the ender3S1/S1 pro in the April 2023 update

LA3QMA, 2023-03-14:
I did not try anymore as i got information on the Discord channel i think that there is a release in near future. The macros i made works ok but then its without any GUI. And the height has to be adjusted manually. So i'll just wait for Crealitys release. Doing "research" i managed to fry one main board.

Creality-Ralf, 2023-05-05:
> > We will support the laser engraving function of the ender3S1/S1 pro in the April 2023 update
> 
> @Creality-Ralf I hope so. I can't wait. Can you tell is the date that the release come?
You can now update your system and use this feature

marcel-1984, 2023-05-05:
> > > We will support the laser engraving function of the ender3S1/S1 pro in the April 2023 update
> 
> > 
> 
> > @Creality-Ralf I hope so. I can't wait. Can you tell is the date that the release come?
> 
> You can now update your system and use this feature
> 
> 

I've already update. I have the function now, but it don't work. I get a failure when I switch to laser when it is connected

Creality-Ralf, 2023-05-05:
> I've already update. I have the function now, but it don't work. I get a failure when I switch to laser when it is connected
What happened？Could you show some photo about that？

marcel-1984, 2023-05-05:
> What happened？Could you show some photo about that？
[image]

Creality-Ralf, 2023-05-05:
> [image]

Click on the bot icon in the upper right corner of the screen, select "Upload log to customer service," and send the serial number of the upload here

marcel-1984, 2023-05-05:
> Click on the bot icon in the upper right corner of the screen, select "Upload log to customer service," and send the serial number of the upload here

Tomorrow I'm back home. Then I will do both options. Thanks till so far

LA3QMA, 2023-05-05:
Turn off the printer.
Disconnect the hotend flat cable
Connect the laser.
Turn on the printer then the SonicPad and quickly change to "laser mode".

Then i can move the x/y axis from the sonicpad.

But engraving doesnt work.
I have made a drawing in Lightburn and when trying to "print" the sonic pad issues these warnings:

{"code":"key61, "msg":"Unknown command:M9", "values": ["M9"]}
{"code":"key61, "msg":"Unknown command:M05", "values": ["M05"]}
{"code":"key61, "msg":"Unknown command:M03", "values": ["M03"]}
{"code":"key272": "msg":"Invalid speed in 'G0 X69.098 Y68.615 F0'", "values":["G0 X69.098 

If i edit the gcode:
Remove all M9 commads
Replace all M05 with M5
Replace all M03 with M3
Remove all F parameters from the G0 commands

Then the laser starts moving to where it should start engraving.
When the laser turn on it stops and the laser is turned off and the following error appears:

MCU 'mcu' shutdown: Scheduled digital out event will exceed max_duration

I have sendt the log from the pad so you already probably have the complete log.

Creality-Ralf, 2023-05-06:
> Turn off the printer. Disconnect the hotend flat cable Connect the laser. Turn on the printer then the SonicPad and quickly change to "laser mode".
>
> Then i can move the x/y axis from the sonicpad. [...]
send the serial number of the upload here or to luojianxiao105776@creality.com

Creality-Ralf, 2023-05-23:
Did you solve the problem?

NayPrints, 2024-03-20:
Apagado de MCU 'mcu': el evento de salida digital programado superará los max_duration 
MISMO FALLO....

elliottcoe, 2024-09-02:
Any updates on this issue? I'm getting the same "failure" message when I try to switch to laser mode.