CrealityOfficial/Creality_Sonic_Pad #7: Customizations for fluidd_pad display

smwoodward, 2022-11-19:
I'm assuming that what is displayed on the sonic pad is based off of KlipperScreen, but i would like to be able to customize what's on the screen some.  Additionally being able to update klipper, mainsail, and fluidd at a faster pace then when updates are released from Creality.  can we get the root credentials to use for customization?

marcribe, 2023-01-02:
Yes, please!

Creality-Ralf, 2023-01-03:
The sonic pad ui is not based on the klipper screen, they are completely different things. If you have a UI request, you can ask for it in the "pull requests" section

marcribe, 2023-01-03:
the visual is exactly the same... including the colors, icons, panels, functions...

Creality-Ralf, 2023-01-09:
I will close this issue and it looks like it has been resolved

closed 2023-01-09 by Creality-Ralf