CrealityOfficial/Creality_Sonic_Pad #8: Only usb1 is suitable for printer connection

samuraev, 2022-11-30:
After update from 29.11.2022 only USB1 could be used for connection of the printer. Other USBs (2,3 and 4) don't detect the printer.

Sebazzz, 2022-12-02:
Thanks @samuraev  - this firmware update left me banging my head against the wall until I found this.

Indeed the printer must now be connected to USB1, which is the top-left USB port. This is not the suggested USB port by Creality, as that is the back-top USB port.

In addition, the serial address has changed from `/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0` to `/dev/serial/by-id/usb_serial_1`.

Sebazzz, 2022-12-03:
Update: Once the Creality Sonic Pad has been set-up and configured you *are* able to use the USB-ports. For instance to back-top port is addressed by `/dev/serial/by-id/usb_serial_3` or `/dev/serial/by-path/platform-5200000.ehci1-controller-usb-0:1.2:1.0-port0`.

CrealityTech, 2023-01-03:
Yes, the update needs to default to using USB1 as the port for the first added model. We mentioned this in the changelog, but unfortunately most users didn't notice it. I think it's because our UI hints weren't clear enough. We will consider using bubble hints in the UI later.

Creality-Ralf, 2023-01-03:
Yes, the update needs to default to using USB1 as the port for the first added model. We mentioned this in the changelog, but unfortunately most users didn't notice it. I think it's because our UI hints weren't clear enough. We will consider using bubble hints in the UI later.

CrealityTech, 2023-01-03:
@samuraev Has the problem been solved?

Escrich, 2023-03-24:
[mcu]  # Serial printer USB conection, Conexión USB de la impresora
# serial: /dev/serial/by-id/usb_serial_1 # Working by default top-left USB port; Izquierda arriba
# serial: /dev/serial/by-id/usb_serial_2 # bottom-left USB port; Izquierda abajo
serial: /dev/serial/by-id/usb_serial_3 # upper-back USB port; Atrás arriba
# serial: /dev/serial/by-id/usb_serial_4 # lower-back USB port; Atrás abajo
restart_method: command

closed 2023-05-23 by Creality-Ralf