CrealityOfficial/Creality_Sonic_Pad #9: Bed leveling option gone and doesnt see resonance sensor from Touchscreen

mottld, 2022-12-12:
As the title says, I no longer see ANY bed leveling buttons under Configure. They used to be there and worked. The touchscreen also can no longer detect the resonance sensor, but input shaping works fine from the console using TEST_RESONANCES AXIS=X and TEST_RESONANCES AXIS=Y commands.

I am on the latest 154 code.`

Creality-Ralf, 2023-01-03:
Have you modified your printer.cfg?

mottld, 2023-01-04:
I moved the following to just before the commented out section at the bottom of the printer.cfg and it resolved the issue with the resonance testing from the pad

[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None
spi_speed: 2000000
spi_bus: spidev2.0

[resonance_tester]
accel_chip: adxl345
accel_per_hz: 70
probe_points:
    117.5,117.5,10

I also had a blank line in the bltouch section. When I deleted the blank line the buttons for bed leveling returned.

Creality-Ralf, 2023-01-09:
I will close this issue and it looks like it has been resolved

closed 2023-01-09 by Creality-Ralf