CrealityOfficial/Creality_Sonic_Pad #11: CR10 Smart Pro

BigBadAl-maker, 2023-01-03:
Hi,when running resonance testing using the Creality G sensor and cable on a stock CR10 Smart Pro, the firmware makes a lot of incorrect changes to the 'Printer.cfg' file that screw it up.

Is there a firmware update of a good copy of the 'Printer.cfg' file?

Also started having a strange unexplained issue that when turning the LED light bar off via the Sonic Pad screen that it turns the printer off.  Not a critical issue.

Creality-Ralf, 2023-01-09:
Version 154 does have problems with vibration compensation, especially in multi-camera situations, and we will fix this in the January update

BigBadAl-maker, 2023-01-17:
New firmware build just released and sorted out both issues.

thanks.

closed 2023-01-17 by BigBadAl-maker