CrealityOfficial/Creality_Sonic_Pad #15: kernel source missing

chealy, 2023-01-22:
I don't see where the Linux kernel source is for this device.  This should be accessible so the device can comply with the GPL.

svobodnik, 2023-02-10:
telling us how their magic SoC works would, like, totally ruin their 3d printer business, man

0xD34D, 2023-03-29:
> telling us how their magic SoC works would, like, totally ruin their 3d printer business, man

Selling this as "open source" and leaving out the source to one of the biggest GPL licenced projects is a misleading business practice.

chealy, 2023-04-21:
@Creality-Ralf 
It's been several months now since this issue was opened.  Please provide an ETA for when Crealty will address this lack of opensource support.

closed 2023-05-23 by Creality-Ralf

chealy, 2023-05-26:
Thanks for sharing the link.  Please correct me if my understanding is wrong, but it appears that the Sonic Pad uses an Allwinner R818 SoC running the 4.9 Linux kernel.  Can you confirm what I'm seeing in the code drop and that this is what the Sonic Pad uses?