CrealityOfficial/Creality_Sonic_Pad #23: LED Light

Toddi3D, 2023-02-12:
I cant see a Button for the LED, why ? Can i add a Button for that ?

Best Regards

Escrich, 2023-03-30:
Yes you can, here I show you how to use a pushbutton , if you a have, or want to have, an output a for a led you simply need to define it, I give you info for input and output:

`[gcode_button][¶](https://www.klipper3d.org/Config_Reference.html#gcode_button)
Execute gcode when a button is pressed or released (or when a pin changes state). You can check the state of the button by using QUERY_BUTTON button=my_gcode_button.

[gcode_button my_gcode_button]
pin:
#   The pin on which the button is connected. This parameter must be
#   provided.
#analog_range:
#   Two comma separated resistances (in Ohms) specifying the minimum
#   and maximum resistance range for the button. If analog_range is
#   provided then the pin must be an analog capable pin. The default
#   is to use digital gpio for the button.
#analog_pullup_resistor:
#   The pullup resistance (in Ohms) when analog_range is specified.
#   The default is 4700 ohms.
#press_gcode:
#   A list of G-Code commands to execute when the button is pressed.
#   G-Code templates are supported. This parameter must be provided.
#release_gcode:
#   A list of G-Code commands to execute when the button is released.
#   G-Code templates are supported. The default is to not run any
#   commands on a button release.`

`[output_pin][¶](https://www.klipper3d.org/Config_Reference.html#output_pin)
Run-time configurable output pins (one may define any number of sections with an "output_pin" prefix). Pins configured here will be setup as output pins and one may modify them at run-time using "SET_PIN PIN=my_pin VALUE=.1" type extended [g-code commands](https://www.klipper3d.org/G-Codes.html#output_pin).

[output_pin my_pin]
pin:
#   The pin to configure as an output. This parameter must be
#   provided.
#pwm: False
#   Set if the output pin should be capable of pulse-width-modulation.
#   If this is true, the value fields should be between 0 and 1; if it
#   is false the value fields should be either 0 or 1. The default is
#   False.
#static_value:
#   If this is set, then the pin is assigned to this value at startup
#   and the pin can not be changed during runtime. A static pin uses
#   slightly less ram in the micro-controller. The default is to use
#   runtime configuration of pins.
#value:
#   The value to initially set the pin to during MCU configuration.
#   The default is 0 (for low voltage).
#shutdown_value:
#   The value to set the pin to on an MCU shutdown event. The default
#   is 0 (for low voltage).
#maximum_mcu_duration:
#   The maximum duration a non-shutdown value may be driven by the MCU
#   without an acknowledge from the host.
#   If host can not keep up with an update, the MCU will shutdown
#   and set all pins to their respective shutdown values.
#   Default: 0 (disabled)
#   Usual values are around 5 seconds.
#cycle_time: 0.100
#   The amount of time (in seconds) per PWM cycle. It is recommended
#   this be 10 milliseconds or greater when using software based PWM.
#   The default is 0.100 seconds for pwm pins.
#hardware_pwm: False
#   Enable this to use hardware PWM instead of software PWM. When
#   using hardware PWM the actual cycle time is constrained by the
#   implementation and may be significantly different than the
#   requested cycle_time. The default is False.
#scale:
#   This parameter can be used to alter how the 'value' and
#   'shutdown_value' parameters are interpreted for pwm pins. If
#   provided, then the 'value' parameter should be between 0.0 and
#   'scale'. This may be useful when configuring a PWM pin that
#   controls a stepper voltage reference. The 'scale' can be set to
#   the equivalent stepper amperage if the PWM were fully enabled, and
#   then the 'value' parameter can be specified using the desired
#   amperage for the stepper. The default is to not scale the 'value'
#   parameter.`

Example for input:
`[gcode_button pulsador]
pin: PG13 # Pin donde está conectado el pulsador

press_gcode: # gcode vacio, hacemos que funcione al soltar, no al pulsar

release_gcode:  # filament unload procedure   
  {% if (printer.print_stats.state != "printing")%} # requires [virtual_sdcard]   
     filament_unload     
  {% else %}
     M117 La maquina está imprimiendo, ahora no se puede descargar!  
     BEEP
     BEEP
     BEEP 
  {% endif %}`

Example for Ourput:

`[output_pin BEEPER_pin]
pin: PD12  # Fan 2
#   Beeper pin. This parameter must be provided.
pwm: True
#   A piezo beeper needs a PWM signal, a DC buzzer doesn't.
value: 0
#   Silent at power on, set to 1 if active low.
shutdown_value: 0
#   Disable at emergency shutdown (no PWM would be available anyway).
cycle_time: 0.001
#   Default PWM frequency : 0.001 = 1ms will give a tone of 1kHz
#   Although not pitch perfect.`

Creality-Ralf, 2023-05-23:
I will close this issue as it appears to have been resolved

closed 2023-05-23 by Creality-Ralf