CrealityOfficial/Creality_Sonic_Pad #24: Cant change the Printername ! Why ?

Toddi3D, 2023-02-16:
Hi All,
i cant change the Name of my Printer. Its a Anycubic Vyper and i put this Lines in the Config above.

####======>Do not modify this configuration
# !Vyper printer model
# printer_size: 245x245x260 Displays information on the Creality Sonic Pad
#####<======Do not modify this configuration

The Name was visible about 3 Seconds, then its again 1unknown !

Why please ???

Best Regards

BakeJake, 2023-03-12:
@Toddi3D 
### Believe you're looking for this!
Its under settings not in the configuration.
[image]

Creality-Ralf, 2023-03-13:
We tested it without problems, please tell me your system version

Creality-Ralf, 2023-03-13:
[image]

BakeJake, 2023-03-13:
> We tested it without problems, please tell me your system version
> [image]

They added it to the configuration file and not in the settings. That is likely the issue, no? 
Since the name of the printer in the configuration is only printed after its generated. There's no way to change the name from within the config file as far as I'm aware.
[Klipper Config Reference](https://www.klipper3d.org/Config_Reference.html)

They are trying to change the name, not find the name of their printer.

Creality-Ralf, 2023-03-13:
If you want to customise the name of your printer on the pad ui, then follow the reply I gave

closed 2023-05-23 by Creality-Ralf

BakeJake, 2023-05-23:
@Creality-Ralf this isn’t fixed. Changing the name in the config doesn’t do anything. I even checked for you. Ender-3 S2 saved in my config does nothing other than just that.. change it in the config. For reference that’s a commented preliminary descriptor. It is not DEFINING any name doing so. They asked how to
change the name no? Therefore this should not be closed, thanks.

Creality-Ralf, 2023-05-23:
> @Creality-Ralf this isn’t fixed. Changing the name in the config doesn’t do anything. I even checked for you. Ender-3 S2 saved in my config does nothing other than just that.. change it in the [...]

Please upload the contents of your "printer.cfg". We have verified that this can be changed several times when you first gave us feedback on this issue. Let's see what the problem is