CrealityOfficial/Creality_Sonic_Pad #28: Feature Request? Tramming Wizzard!

BakeJake, 2023-03-11:
Is it possible to get some sort of macro, or feature that returns the reading from an ABL on the corners to tram it in?
Like how I was doing with mriscoc professional firmware? [Tramming Menu from Mriscoc](https://github.com/mriscoc/Ender3V2S1/wiki/Bed-tramming-menu#tramming-wizard)

Creality-Ralf, 2023-03-13:
Why do you need this feature?

BakeJake, 2023-03-13:
> Why do you need this feature?

I cannot get it perfect, but I could with that. Yeah my S1 has ABL but it doesn't seem to compensate exactly correct. If its not in perfect tram I can see the difference in height on the first layer and for the piece of mind of knowing its on par!

Creality-Ralf, 2023-03-13:
Okay, we'll think about it.

BakeJake, 2023-03-13:
> Okay, we'll think about it.

Thank you very much!

derTimm0, 2023-03-13:
why you dont use the "bed screws adjust" feature on klipper via web-gui?

BakeJake, 2023-03-13:
> why you dont use the "bed screws adjust" feature on klipper via web-gui?

Yeah, I've thought about that! I didn't try yet! I hope it shows the numbers then I wont be too concerned, but through the web-gui only? cant that be done from like advanced options on the device as well?

derTimm0, 2023-03-13:
the sonic pad use a own Frontend. on "normal" klipperscreen display gui it will be displayed. we must hope on "creality" that he implate this feature.

BakeJake, 2023-03-13:
Thank you for that!

Creality-Ralf, 2023-03-22:
It looks like the problem has been solved. Then I won't consider adding this UI for now

BakeJake, 2023-03-26:
> It looks like the problem has been solved. Then I won't consider adding this UI for now

The problem hasn't been solved! I've tried to do what was suggested but that just moves the toolhead to the corners to manually set the springs but doesn't probe to give a value. Id like to be able to Dial it in!

BakeJake, 2023-05-06:
@Creality-Ralf I was able to achieve the same results by using "SCREWS_TILT_CALCULATE", except it doesn't have any visualization and its all in the console! would be nice for one inside the Sonic Pads Firmware!

hweg2007, 2023-06-22:
> Why do you need this feature?

Because it makes it 10 times easier to level the bed than having to manually do it with paper and end up giving up cuz it’s annoying