CrealityOfficial/Creality_Sonic_Pad #29: Cr-x pro

Brekeena01, 2023-03-13:
Any idea when support will be added out if the box for the CR-X pro?

Creality-Ralf, 2023-03-13:
The CR-X is already in the renewal programme for this year. However, it is not possible to say exactly when the update will take place. You can try following the tutorial on our youtube channel to configure

Brekeena01, 2023-03-14:
Ok. Thanks. Super keen to get it running as I have the sonic pad sitting here waiting. I don’t think I’m confident enough in my ability to configure it manually. Thanks for the update.

Brekeena01, 2023-03-14:
I would actually probably have a go but I can’t find a config file for the cr-x pro at all.

Creality-Ralf, 2023-03-22:
https://www.reddit.com/r/3Dprinting/comments/ci6cdb/finally_running_klipper_on_a_creality_crx_hope/

Creality-Ralf, 2023-05-23:
It looks like the problem has been solved and I will close the issue

closed 2023-05-23 by Creality-Ralf