CrealityOfficial/Creality_Sonic_Pad #30: M600 for color change and Sonic pad screen wake up on events

Escrich, 2023-03-19:
Dear all, if you try to make a colour change using M600 from slicer you have a problem, the answer from Sonic pad is poor, we need to be free to change your lately added M600 by our tailored ones, this is not a whim, it’s a real need.
Just being different it’s really related to this, and some more other things, the screen, MUST wake up everything something happens, if you have a filament runout, a pause or whatever you could have, and the sonic pad screen is hibernating, the screen must wake up to show the related information, in some cases is not only information, it’s just waiting for a on screen button to be pressed, but you can’t see it, because the screen is sleeping

Escrich, 2023-03-19:
I was testing something now, and firmware code M600 can be overriden this way :

[gcode_macro M600]
rename_existing: M600.1
gcode:
   NOTIFY TEXT="Empieza el cambio de color automatico" CLEAR=0
   BEEP
   BEEP
   M118 Cambio de color automático en marcha
   color

Escrich, 2023-05-03:
Still no answers?????

Creality-Ralf, 2023-05-23:
Thanks for the feedback, we plan to add an optional option in a subsequent version: "The screen will be automatically woken up in case of an error"

Escrich, 2023-05-23:
may be not only in the case of an error, just every time something happens, or everytime M117 or M118 are executed, and may be in some more cases too

Escrich, 2023-07-11:
any news here?

SurfingSpider, 2023-07-11:
I would love to see this too.

Creality-Ralf, 2023-07-13:
This feature is already in testing. Integration is planned for the August update

Escrich, 2023-09-17:
I think you could do it better

Creality-Ralf, 2023-09-18:
> I think you could do it better

Now it's available. Looking forward to your feedback.

Escrich, 2023-10-23:
not enough

Creality-Ralf, 2023-10-30:
> not enough

Please describe your idea in detail.