CrealityOfficial/Creality_Sonic_Pad #32: Bulk files back up

Escrich, 2023-03-24:
From my point of view it's a must, to have the possibility to download, from graphic environment, all , or some configuration files at a time, instead to download them one by one

Escrich, 2023-05-03:
?????????????, no answer???

Creality-Ralf, 2023-05-23:
Not quite understand what you mean, can you describe it with a picture?

Escrich, 2023-05-23:
It's so simple, everytime I need to make a backup, I need to download the Sonic Pad's configuration files, one by one, just looking at other machines I own, I could be able to download, for backup purposes, or any other reason, the amount of files and / or folders I want, I need to do the same at Sonic Pad, enclosed I show you two pictures, the Sonic Pad No bulk download, and another machine, no Sonic pad, bulk download

Bulk Dowload picture, you could see the selection squares and the cloud to download the whole selected items, doesn't matter if the selected item is a file or a folder

[image]

Sonic Pad No bulk download picture, you could see the dialog to download, referring a single previously selected file, and no more, you need to do this process file by file, because even not folders are allowed to be downloaded

[image]

Creality-Ralf, 2023-07-13:
The first picture you posted is mainsail and the second picture is fluidd, they are different programs. You can get to our preconfigured mainsail by typing ":8819" after the IP address of the sonicpad. in mainsail you can find this function in the corresponding position

Escrich, 2023-07-17:
I know both Fluidd and Mainsail, and I know how to handle each on the Sonic Pad, the Mainsail screen comes from a machine, other than the sonic pad, really a mini PC, just to show you what we need, because this functionality has not been yet implemented in Sonic pad, both in fluid neither mainsail

Creality-Ralf, 2023-07-18:
> I know both Fluidd and Mainsail, and I know how to handle each on the Sonic Pad, the Mainsail screen comes from a machine, other than the sonic pad, really a mini PC, just to show you what we need, because this functionality has not been yet implemented in Sonic pad, both in fluid neither mainsail
"The first picture you posted is mainsail and the second picture is fluidd, they are different programs. You can get to our preconfigured mainsail by typing ":8819" after the IP address of the sonicpad. in mainsail you can find this function in the corresponding position"
I'm on version 1.0.6.49.145 and it works, please upgrade to the latest version and try again, also you need to select any file for the icon to appear!