CrealityOfficial/Creality_Sonic_Pad #35: No Downgrade possible ???

Toddi3D, 2023-03-24:
It would be great if we can do a Downgrade to previous Firmware.

Creality-Ralf, 2023-05-23:
What specific problems did you have that required a downgrade? In fact, we have released a USB burner package to recover sonicpad that has crashed for some reason

TheJadeGeek, 2023-05-26:
Unable to level when bed is heated, get probe error or end stop error after a few probes. 

setting Z Offset appears to do nothing, my bed is now scratched all to hell. 

just a couple issues with the most recent update that would warrant wanting to rollback a version.

Creality-Ralf, 2023-05-29:
> Unable to level when bed is heated, get probe error or end stop error after a few probes.
> 
> setting Z Offset appears to do nothing, my bed is now scratched all to hell.
> 
> just a couple issues with the most recent update that would warrant wanting to rollback a version.

Try using this: https://github.com/CrealityOfficial/Creality_Sonic_Pad_Firmware

TheJadeGeek, 2023-05-29:
Yeah I eventually got that to work. After lots of help. Found out that my issue is a factory defect with my cr10 smart and a factory spliced cable failed.