CrealityOfficial/Creality_Sonic_Pad #40: Sound notifications on Sonic pad?

BakeJake, 2023-04-30:
It would be awesome if we can get something like M300, for the sonic pad so that when a print is done, or hot we can say play different keytones to notify the user of what's going on, when not exactly next to the printer! I don't knopw the difficulty of adding something like this but you should be able to control via gpio or something right? Would be really nice!

Escrich, 2023-05-03:
I did it, but using a couple of simple macros, and one output pin on my Btt Octopus card

BakeJake, 2023-05-03:
> 

So I have the Ender 3 S1 with stock "F1" board, and It does have a buzzer in the display I think, not sure about on the board.
How would I be able to find the pin name or all of that? Can I access accessories of the printer with klipper? Can we also turn on and off the LED lightbar upgrade or is that not possible?

Freddyb888, 2023-05-04:
> I did it, but using a couple of simple macros, and one output pin on my Btt Octopus card

How exactly, I have the btt octopus max ez. Thx

BakeJake, 2023-05-07:
> > I did it, but using a couple of simple macros, and one output pin on my Btt Octopus card
> 
> How exactly, I have the btt octopus max ez. Thx

Wondering the same!

Escrich, 2023-05-10:
`# **********************************************************************************
# Beeper definition
# **********************************************************************************

[output_pin BEEPER_pin]
pin: PD12  # Fan 2 # must be changed acording to your board, or the pin you choose, in this case is good to use a 5V fan output with a 32 Ohm speaker
#   Beeper pin. This parameter must be provided.
pwm: True
#   A piezo beeper needs a PWM signal, a DC buzzer doesn't.
value: 0
#   Silent at power on, set to 1 if active low.
shutdown_value: 0
#   Disable at emergency shutdown (no PWM would be available anyway).
cycle_time: 0.001
#   Default PWM frequency : 0.001 = 1ms will give a tone of 1kHz
#   Although not pitch perfect.

# **********************************************************************************
[gcode_macro M300]
description: Gcode to generate sound on the speaker or PASIVE buzzer
gcode:
    # Use a default 1kHz tone if S is omitted.
    {% set S = params.S|default(880)|int %}
    # Use a 10ms duration is P is omitted.
    {% set P = params.P|default(300)|int %}
    SET_PIN PIN=BEEPER_pin VALUE=0.5 CYCLE_TIME={ 1.0/S if S > 0 else 1 }
    #SET_PIN PIN=BEEPER_pin VALUE=0.5 CYCLE_TIME=0.0011363
    #G4 P{P}
    SET_PIN PIN=BEEPER_pin VALUE=0

# **********************************************************************************

[gcode_macro BEEP]
description: Makes a simple beep sound.
gcode:
    M300 P200 S880

`

chuckxxxl, 2023-05-19:
Sonic Pad cant make beep

Escrich, 2023-05-19:
You should make beep outside the Sonic pad, I have a simple 32 ohm speaker connected to a fan output, look in your board for a place to connect the speaker

Escrich, 2023-07-11:
Second part of the year started on July the first, we´re expecting