CrealityOfficial/Creality_Sonic_Pad #41: News Update ist Out but..... Where ist the Manual for Laser

chuckxxxl, 2023-05-01:
the new update is finally out. there is now the laser function in the sonicpad. how the fuck does that work? where is the manual please? what software is required? Do you need the control box (falcon) for the laser or is the laser enough? I've tested it with different software, constantly error messages.

Creality-Ralf, 2023-05-23:
Laser data files need to be obtained using creality print, a video tutorial will be available soon

thomazff, 2023-05-28:
That's a shame. Creality app is too simplistic. 

There has to be lightburn suport.

Creality-Ralf, 2023-05-29:
> That's a shame. Creality app is too simplistic.
> 
> There has to be lightburn suport.

Try using this: https://github.com/CrealityOfficial/Creality_Sonic_Pad_Firmware

thomazff, 2023-05-29:
> What software do you want to use

Lightburn 

Didn't get your second message, you sent me to a firmware burning tool. This is about laser engraving.

Creality-Ralf, 2023-06-30:
> > What software do you want to use
> 
> Lightburn
> 
> Didn't get your second message, you sent me to a firmware burning tool. This is about laser engraving.

Ok, we will make this software available