CrealityOfficial/Creality_Sonic_Pad #42: LPC1769

parogc, 2023-05-01:
Is the sonic pad compatible with LPC1769 MCU? where is the compatibility list?

Nanodesigner, 2023-05-20:
ho lo stesso problema perchè la skr 1.4 turbo ha un LPC1769. si è trovata una soluzione?

Creality-Ralf, 2023-05-23:
LPC's PAD UI compilation support will be available in June

BloodSteyn, 2023-07-27:
> LPC's PAD UI compilation support will be available in June

Any news on this? I just got the Sonic Pad and have an LPC1768 that I couldn't compile for, and the Pad said it is on the latest version.

Creality-Ralf, 2023-08-05:
> > LPC's PAD UI compilation support will be available in June
> 
> Any news on this? I just got the Sonic Pad and have an LPC1768 that I couldn't compile for, and the Pad said it is on the latest version.

LPC 1768 and LPC 1769 are already supported on version V1.0.6.49.145.Please check your version. If you don't get an OTA alert, you can try restarting the pad, or go here (https://www.creality.com/pages/download-creality-sonic-pad?spm=. .page_1934481.products_display_1.1&spm_prev=. .page_1937021.header_1.1) to download the upgrade package!

closed 2023-09-01 by Creality-Ralf