CrealityOfficial/Creality_Sonic_Pad #43: This feature from klipperscreen would be awesome and greatly appreciate by many im sure.

BakeJake, 2023-05-07:
https://klipperscreen.readthedocs.io/en/latest/Quicktips
Scroll down to **Layer Progress**, and why do we not have the latest version or all the features of klipper?
We don't have object cancelation? that's the only one I can think of off top of my head. but being able to
have cura or prusaslicer communicate with the screen could make things a little more accurate!

blazecoding2009, 2023-05-07:
Haven't tried it yet but you can do smtg like this in prusa slicer or its derivatives

in ur start gcode add: ```;LAYER_COUNT:[total_layer_count]```
and before layer change: ```;LAYER={layer_num + 1}```

BakeJake, 2023-05-07:
> Haven't tried it yet but you can do smtg like this in prusa slicer or its derivatives
> 
> in ur start gcode add: `;LAYER_COUNT:[total_layer_count]` and before layer change: `;LAYER={layer_num + 1}`

Thank you for this, do you think thats why this isnt working? my naming is unconventional? 

SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
; total_layer_count = {total_layer_count}

The print stats I had to comment out because it doesn't work with sonicpad some reason.

Its only on certain prints with like ironing/variable height that the layers are way off! Ill have to try changing it to LAYER_COUNT but pretty sure I tried everything I found lol! :|

Creality-Ralf, 2023-05-23:
This feature will be available in the near future with the built-in klipper version upgrade

BakeJake, 2023-05-23:
> This feature will be available in the near future with the built-in klipper version upgrade

Glad to hear! Does that mean we’ll get the latest klipper? I’d love to use object exclude. As well as be able to use git in the sonic pad, just provide a factory backup you know? And let us put what we want since we paid for the whole unit? The people that are more advanced that is would be able to flash whatever they wanted. As well as have control over fluidd or mainsail where we can’t even access updates because it wants us on a strict path! I’d like freedom, I didn’t spend almost $200 to not be able
To Use klipper to its full potential! Open source is open source, not locked open source! That’s not open lol

BakeJake, 2023-05-23:
> This feature will be available in the near future with the built-in klipper version upgrade

Things like adaptive meshes and just smart adding like this are impossible with the state you left it in.

blazecoding2009, 2023-05-23:
Also the thumbnails for the print files don't come up on mainsail. Kinda weird.

BakeJake, 2023-05-23:
> Also the thumbnails for the print files don't come up on mainsail. Kinda weird.

They come up on mine, it REALLY matered the order you put them too. You want thumbnail before the full image, I use 32or64 for thumbnail and 300-320 for other. Haven’t had any issues displaying on both, but I’m curious how your concluding this issue!

BakeJake, 2023-05-23:
> Haha here's the thing, It is like that
> [image]

Just for the heck of it can you try putting that to 300x300 see if it works? Is that superslicer? I’ve had luck with prusaslicer(which I switched to) cura, and super slicer! So it should def work!

BakeJake, 2023-05-23:
> It's showing up on the screen, it's mainsail which is the issue. And no, its Orca Slicer

Oh okay, never even heard of that! That might be the issue that it’s integration of thumbnails is not the same! I know prusaslicer let’s you select jpeg, and png but I’m pretty sure jpeg doesn’t show up! So be careful!

blazecoding2009, 2023-05-23:
Yeah pretty weird. I'll check

Creality-Ralf, 2023-05-23:
> > This feature will be available in the near future with the built-in klipper version upgrade
> 
> Things like adaptive meshes and just smart addons like this are impossible with the state you left it in.

Yes, this feature will require you to operate on the screen UI. We will consider making this an automatic operation in the future

Creality-Ralf, 2023-05-23:
> Found a way to fix it, you gotta add
> 
> ```
> ; generated by SuperSlicer 1.5.0 on 2023-04-21 at 13:13:47
> ```
> 
> in your start gcode

Tks.This issue will be fixed in the near future.

blazecoding2009, 2023-05-23:
Okay also it seems that after adding that line, None of my prints are working. They are immediately paused. This is what it says in the klipper logs: ```
[ERROR] 2023-05-23 11:21:57,935 [root] [virtual_sdcard:_load_file:251] File opened:3DBenchy_PLA_1h31m.gcode Size:40 start_print
[ERROR] 2023-05-23 11:21:58,781 [root] [virtual_sdcard:work_handler:757] virtual_sdcard dispatch
Traceback (most recent call last):
  File "/usr/share/klipper/klippy/extras/virtual_sdcard.py", line 657, in work_handler
    gcode_move.recordPrintFileName(print_file_name_save_path, self.current_file.name, fan_state=self.fan_state, filament_used=self.print_stats.filament_used, last_print_duration=self.print_stats.print_duration)
AttributeError: PrintStats instance has no attribute 'print_duration'
[ERROR] 2023-05-23 11:21:58,785 [root] [virtual_sdcard:work_handler:798] filename:/root/gcode_files/3DBenchy_PLA_1h31m.gcode end print
[ERROR] 2023-05-23 11:23:06,729 [root] [__init__:logging_callback:267] Got EOF when reading from device```

blazecoding2009, 2023-06-10:
@Creality-Ralf Any updates on this?

BakeJake, 2023-06-10:
> Okay also it seems that after adding that line, None of my prints are working. They are immediately paused. This is what it says in the klipper logs: ` [ERROR] 2023-05-23 11:21:57,935 [root] [...]

This looks liked you entered the wrong moonraker info and tried uploading a gcode file?
I found out with orca slicer klipper doesn't scan for it. It does now, but that's when these guys decide were worth the new versions :D I ran into that trying orca! its not orcas fault lol

blazecoding2009, 2023-06-11:
Yeah for whatever reason it was bugging out. Completely forgot what I did to fix it

blazecoding2009, 2023-06-11:
Wondering when they'll add the new features

blazecoding2009, 2023-06-11:
It wouldn't be too hard if they had open sourced it

BakeJake, 2023-06-11:
> It wouldn't be too hard if they had open sourced it

With ya on that, would be nice to even have the device unlocked with a "recovery" image or little program that resets it you know?
so we could test things, and always go back.
You might have just added the port or something to the printer instead of the main ip to fix it?
or deleted the date off the superslicer comment?

blazecoding2009, 2023-06-11:
There is a command to reset it, you gotta ssh tho, its ```/usr/share/script/recovery.sh all```

BakeJake, 2023-06-11:
> There is a command to reset it, you gotta ssh tho, its `/usr/share/script/recovery.sh all`

I love the 3d printing community for little things like this <3 lol but yeah, if that's the case, unlocking sudo and whatnot would be a godsend!

blazecoding2009, 2023-06-11:
Yeah

Creality-Ralf, 2023-06-30:
This feature is supported in the June update and you can operate it directly on the pad screen, please make sure you slice with the correct settings