CrealityOfficial/Creality_Sonic_Pad #44: Creality PAD outdated.

schmittm63, 2023-05-08:
Some Klipper and Moonraker features will not work because the Klipper/Moonraker version that PAD is based on is heavily outdated klipper release. For example, the Creality PAD doesn't get along with the Orca Slicer (see https://github.com/SoftFever/OrcaSlicer/issues/840).
Also new functions like "EXCLUDE OBJECT" still missing.
Please FIX this ASAP.

Creality-Ralf, 2023-05-23:
The built-in version of klipper will be updated to V0.11 in the near future

Martin2112, 2023-06-17:
Can you also update Moonraker to a more recent version and include the 'notifier' plugin and it's dependencies. Thanks.

Creality-Ralf, 2023-06-30:
The components have been updated in the June update, please check

vser, 2025-01-04:
@Creality-Ralf can you push the latest update changes to Github please? It is so tedious not having these settings/configurations to change easily.