CrealityOfficial/Creality_Sonic_Pad #53: print out of range

vinaysamudrala, 2023-06-05:
Getting print out of range error with Ender 3S-1 pro in sonic pad after the print is getting completed.
List of errors after completion of print:
1. All the 3 axes motors are staying in the same position with motors in the lock position just above the print.
2. Hot end and heat bed are staying hot even after the print completes. (causing extruder jam leading expanding of PLA filament at the inlet PTFE tube of the extruder)
3. Videos are unable to save after the completion of the print.
4. Showing an **out-of-range** error after the print.

beroliv, 2023-07-23:
pls check if your slicer is set to the same bed size like your printer! if your slicer is set to big then you will have this issue!

vinaysamudrala, 2023-07-31:
> pls check if your slicer is set to the same bed size like your printer! if your slicer is set to big then you will have this issue!

Tq,
To utilize max volume I have changed some settings.

Issue solved.

BradGillies, 2023-08-29:
I have this same issue.  It started after the last softrware update on the sonic pad.  Parameters for the printer have not changed but  am getting this error after every print.
Mine is an ender 3V2 with the Sprite pro extruder and hotend.  I am not sure which axis is going out of range but the print head is well within range when it stops.