CrealityOfficial/Creality_Sonic_Pad #65: Latest firmware no longer allows printing from sonic pad interface

FilthyMick, 2023-07-08:
I have two printers connected to the Sonic Pad. Both worked well until the last update(s). After update, I could not send prints from the pad to the printers. I could print from Fluidd interface as normal, and no other communication issues are present from the pad to the machines. I reflashed both printers, but only one has returned to normal state. 

When selecting a file to print on my Ender 5 Plus with BTT SKR Mini E3 V3 board, the first message "Loading", is followed immediately by "Failure!". I can otherwise control the machine fine from the pad, but can only send prints from the web interface. Nothing is visible in the logs, as though the pad never attempted to send the file.

Sacrasang, 2023-07-19:
Bump, this issue has also been present on my pad since updating to latest firmware.

melt-it, 2023-07-30:
Hi, I have the same problem.
Even a reset to factory settings did not bring any success.
However, the problem only exists when I want to start printing from the pad.

DMcL-54, 2024-01-10:
Me too since a firmware update the sonic pad is stuck on a boot screen loop