CrealityOfficial/Creality_Sonic_Pad #72: Please add a printer.cfg for Ender 3 PRO 4.2.7 with CRTOUCH and SPRITE PRO

hardlevel, 2023-07-30:
The printer.cfg available use the normal extruser, please add a config to use with sprite pro!

I tried to change the rotation_distance: 7.5 but It is doing under extrusion, there's something elase that must be configured?

beroliv, 2023-07-31:
change the setting in your printer.cfg. an not over the Gui in Klipper Screen

CrealityTech, 2023-08-03:
> The printer.cfg available use the normal extruser, please add a config to use with sprite pro!
> 
> I tried to change the rotation_distance: 7.5 but It is doing under extrusion, there's something elase that must be configured?

Just change rotation_distance;  sprite paragramed is:424.9 per/mm     
the distance is: 360/1.8*16/424.9