CrealityOfficial/Creality_Sonic_Pad #74: Blue covering on being printed part does not appear anymore

Escrich, 2023-08-05:
In the begining, before the latest two or three firmware updates, we had a blue cover coming in the image from bottom to top, depending on the printed percentage, currently, you can print a part but you can not see how it is covered in blue, will it be restored again?

kronas0, 2023-09-11:
> In the begining, before the latest two or three firmware updates, we had a blue cover coming in the image from bottom to top, depending on the printed percentage, currently, you can print a part but you can not see how it is covered in blue, will it be restored again?

Have you changed the setting of Cura Slicer?   This function base on the Slicer setting.

Escrich, 2023-09-11:
I never used Cura slicer, it was working using Prusa slicer, without any problem