CrealityOfficial/Creality_Sonic_Pad #76: Self test for Ender5 S1 failed

qnub, 2023-08-19:
Self test for Ender5 S1 failed with error `{'error': 'WebRequestError', 'message': "'NoneType' object has no attribute 'isenable'"}`, right after it it show message `Internal error on command:BED_MESH_SET_DISABLE`

homeveg, 2023-08-19:
I have precisely the same error! I got my SonicPad just today. The printer itself was running the latest factory firmware (V1.0.7)

kronas0, 2023-09-11:
> Self test for Ender5 S1 failed with error `{'error': 'WebRequestError', 'message': "'NoneType' object has no attribute 'isenable'"}`, right after it it show message `Internal error on command:BED_MESH_SET_DISABLE`

This error should be fixed while you update the firmware to V1.0.6.51.139.