CrealityOfficial/Creality_Sonic_Pad #87: Please update Moonraker

ale-ben, 2023-11-15:
Please update Moonraker to the latest version to allow support for new config items.

Tried adding [spoolman] to moonraker config but it gives an error. Turns out that the version sonic pad has is to old to support this (and a lot of other) plugins.

This is the official documentation of Moonraker for Spoolman, please update so that we can actually use it...
https://moonraker.readthedocs.io/en/latest/configuration/#spoolman

randybb, 2023-12-11:
The best would be to get rid of this modified klipper and other stuff and go with the way how BIGTREETECH Pad 7 works - just an ordinary "rpi" machine with vanilla klipper. Everything extra can be handled by plugins.

ale-ben, 2023-12-11:
Yeah, definitely!
There are some projects that completely wipe the SP to install debian and obtain that but it's a bit of an overkill for my needs...

KLXYinc, 2023-12-15:
would the screen still work?

ale-ben, 2023-12-17:
@KLXYinc screen relies on api provided by Moonraker, theoretically, screen UI should work without a problem (if they did not tamper to much with the original API).

Besides that, since they are already using open-source products without respecting their license (see #82), they could also switch to Klipper-screen, the most used UI for touch-screen-enabled Klipper devices.

KLXYinc, 2023-12-23:
i have switched to debian with klipper screen now it works better but i miss the creality screen... is there somewhere only the screen for it so i could use it instead of klipper screen?