CrealityOfficial/Creality_Sonic_Pad #95: Update this repository with latest fixes

bistory, 2023-12-29:
Hi,

Could you please update this repository with latest fixes and additions (like Ender 3 V3 SE) ?

Thanks !

vser, 2024-09-16:
_fingers crossed 🤞_ - they finally uploaded almost all of the [Marlin configuration files](https://github.com/CrealityOfficial/Ender-3V3-SE/tree/main). Hopefully only a short wait until they notice this repo again. 🤣

randybb, 2024-09-16:
I bought Sonic Pad because they released updates every month. Not with the latest klipper, but at least there were some updates. This was in december...so after that they stopped releasing updates :( It is working fine, but I like to have everything updated, so finger crossed...