CrealityOfficial/Creality_Sonic_Pad #105: Connecting Spoolman to Sonic Pad

laufantn, 2024-03-28:
I've been trying to connect Spoolman with Moonraker (Fluidd) for days. Unfortunately, I always get an error message after restarting Klipper.

`#################################################################
All Configuration Files:
/mnt/UDISK/printer_config2/moonraker.conf
#################################################################
[ERROR] 2024-03-27 16:36:28,287 [root] [database:__init__:120] ##########/mnt/UDISK/.moonraker_database
[ERROR] 2024-03-27 22:10:58,793 [root] [database:__init__:120] ##########/mnt/UDISK/.moonraker_database
[ERROR] 2024-03-27 22:10:58,949 [root] [server:load_component:283] Unable to load component: (spoolman)
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/server.py", line 275, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'moonraker.components.spoolman'
[ERROR] 2024-03-27 22:23:52,773 [root] [database:__init__:120] ##########/mnt/UDISK/.moonraker_database
[ERROR] 2024-03-27 22:23:52,932 [root] [server:load_component:283] Unable to load component: (spoolman)
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/server.py", line 275, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'moonraker.components.spoolman'
[ERROR] 2024-03-27 22:25:03,581 [root] [common:build_error:575] JSON-RPC Request Error: 503
Klippy Host not connected
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/common.py", line 541, in execute_method
    result = await callback(params)
  File "/usr/share/moonraker/moonraker/websockets.py", line 115, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/usr/share/moonraker/moonraker/klippy_connection.py", line 721, in request
    raise ServerError("Klippy Host not connected", 503)
moonraker.utils.ServerError: Klippy Host not connected
[ERROR] 2024-03-27 22:25:06,238 [root] [common:build_error:575] JSON-RPC Request Error: 503
Klippy Host not connected
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/common.py", line 541, in execute_method
    result = await callback(params)
  File "/usr/share/moonraker/moonraker/websockets.py", line 115, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/usr/share/moonraker/moonraker/klippy_connection.py", line 721, in request
    raise ServerError("Klippy Host not connected", 503)
moonraker.utils.ServerError: Klippy Host not connected
[ERROR] 2024-03-27 22:25:53,687 [root] [database:__init__:120] ##########/mnt/UDISK/.moonraker_database
[ERROR] 2024-03-27 22:25:53,850 [root] [server:load_component:283] Unable to load component: (spoolman)
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/server.py", line 275, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'moonraker.components.spoolman'
[ERROR] 2024-03-27 22:26:28,325 [root] [database:__init__:120] ##########/mnt/UDISK/.moonraker_database
[ERROR] 2024-03-27 22:26:28,497 [root] [server:load_component:283] Unable to load component: (spoolman)
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/server.py", line 275, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
[... 278 lines cut ...]
config_file: /mnt/UDISK//printer_config2/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
log_file: /mnt/UDISK/printer_logs2/moonraker.log
software_version: ?
python_version: 3.7.8 (default, Jun 06 2023, 09:42:28)  [GCC 6.4.1]
[INFO] 2024-03-27 22:55:43,682 [root] [loghelper:__init__:79] data_path: /mnt/UDISK/.moonraker_database
[INFO] 2024-03-27 22:55:43,684 [root] [loghelper:__init__:79] is_default_data_path: False
[INFO] 2024-03-27 22:55:43,685 [root] [loghelper:__init__:79] config_file: /mnt/UDISK//printer_config2/moonraker.conf
[INFO] 2024-03-27 22:55:43,685 [root] [loghelper:__init__:79] startup_warnings: []
[INFO] 2024-03-27 22:55:43,685 [root] [loghelper:__init__:79] verbose: False
[INFO] 2024-03-27 22:55:43,685 [root] [loghelper:__init__:79] debug: False
[INFO] 2024-03-27 22:55:43,685 [root] [loghelper:__init__:79] asyncio_debug: False
[INFO] 2024-03-27 22:55:43,686 [root] [loghelper:__init__:79] is_backup_config: False
[INFO] 2024-03-27 22:55:43,687 [root] [loghelper:__init__:79] is_python_package: False
[INFO] 2024-03-27 22:55:43,687 [root] [loghelper:__init__:79] log_file: /mnt/UDISK/printer_logs2/moonraker.log
[INFO] 2024-03-27 22:55:43,687 [root] [loghelper:__init__:79] software_version: ?
[INFO] 2024-03-27 22:55:43,687 [root] [loghelper:__init__:79] python_version: 3.7.8 (default, Jun 06 2023, 09:42:28)  [GCC 6.4.1]
[INFO] 2024-03-27 22:55:43,740 [root] [confighelper:read_file:1064] Configuration File '/mnt/UDISK/printer_config2/moonraker.conf' parsed, total size: 1711 B
[INFO] 2024-03-27 22:55:43,743 [root] [server:add_log_rollover_item:202] 
#################### Moonraker Configuration ####################`

m3po-mike, 2024-03-30:
Also trying to get Spoolman working with Sonic Pad, but no luck.
I gave up and removed Spoolman
I tried it with 3D-MES and that is working without any problem (no changes to the Sonic Pad needed, works without changing any file), Must say it does not look so nice (yet) as Spoolman.
Maybe something for Spoolman developers to look at to get Spoolman working for all people (including Sonic Pad users).

laufantn, 2024-03-30:
Yes i currently use 3d-mes but i would still like to be able to use spoolman as well