CrealityOfficial/Creality_Sonic_Pad #110: BUG** Firmware does not generate if Communication interface is not set to USB

rocket59, 2024-04-22:
Been trying to generate firmware from the interface for a non creality printer but the firmware will not generate (actually it's stuck on the "generating firmware" message) if the interface is set to serial rather than USB. 

Seems like a major bug which has still not been addressed (I saw a couple of reports on reddit).

rocket59, 2024-04-25:
Does anyone from Creality actually read these issues?

randybb, 2024-05-02:
nope. and since there was no new version since January (previously they released a new version every 1-2 months) so hard to say if they are going to support it. In any case the klipper is pretty old, the best option is to move to BTT Pad, or just move to debian with a vanilla klipper.

rocket59, 2024-05-05:
Solved my issue by SSHing into the pad and doing it via the CLI. Will make me think twice about supporting them for future purchases if they don't support their customers.

Yeah I'm so tempted to put debian on it even though it would be a cut down version.